dl1ycf/pihpsdr #28: RenewRepo #2

qwgsdy, 2024-05-09:
Another attempt also fails.
I am not a LINUX expert. See the attachments.
Please suggest a solution ...
[image]
[image]
[image]
[image]

CT1IQI, 2024-05-10:
To update the local copies of the github repositories you have to give a 'git pull' command from the directory where the local copy is situated. They are under /home/pi/github in 
/home/pi/github/pihpsdr and /home/pi/github/Saturn
But that is not where you are looking, and the error message is crystal clear "Datei nicht gefunden".
So, e.g. to make the latest pihpsdr, do
  cd /home/pi/github/pihpsdr
  git pull
  make clean
  make
You can also put this on one line with && in between the parts which is a logical expression meaning do the following if the previous finished with success.
cd /home/pi/github/pihpsdr && git pull && make clean && make

Confusing in naming is that when all went successful the executable binary will also be called pihpsdr
I normally copy that binary to /usr/local/bin/ directory:
sudo cp /home/pi/github/pihpsdr/pihpsdr /usr/local/bin/

The original OS image does not have a directory /home/pi/pihpsdr. You better remove it after having copied the configuration files it may have to /home/pi/.config/pihpsdr
Looking at the source code of pihpsdr it looks for that directory /home/pi/.config/pihpsdr for its configuration files.
The original OS is not properly configured, has the .config, but not its subdirectory pihpsdr.
You don't see the .config in the GUI because you have not set hidden directories to be shown (the . makes it hidden type)

You can make the directory with 
mkdir -p  /home/pi/.config/pihpsdr
(from the command line you can get loads of help about all commands by typing 'man <command>'; e.g. man mkdir will tell you that the -p option will create intermediate directory levels as needed)

qwgsdy, 2024-05-13:
MARTINUS, Your posting was very helpful. Thank you

dl1ycf, 2024-05-15:
The real problem was that the "git  clone" terminated with some obscure error message.
NOTE FOR G2: it seems that your pihpsdr normally resides in /home/Github/pihpsdr not in /home/pihpsdr.
This then leads to two installations this shoudl be avoided. I shall update the PDF.

dl1ycf, 2024-05-15:
Can this be closed now?

qwgsdy, 2024-05-27:
Yes, it can closed.

closed 2024-05-29 by dl1ycf