dl1ycf/pihpsdr #29: Is it possible to implement the CESSB switch? It doesn't have to be now

qwgsdy, 2024-07-15:
The WDSP manual defines an on/off software switch for the CESSB overshoot control  through the command 'void SetTXAosctrlRun (int channel, int run)'.
In Pihpsdr this switch is set inactive. Add it with default 'off' as a checkbox in the tx menu that is operable when the other conditions are met, compressor active, and linear phase filter chosen.

dl1ycf, 2024-07-29:
I think most things should be automatic, not worrying the end user with too many options. If this control is automatically
set "on" whenever compression is on and the TX filter is linear phase, would this be an improvement?

qwgsdy, 2024-07-29:
It would definitely be an improvement. I would really like to see this feature.

closed 2024-07-29 by qwgsdy