dl1ycf/pihpsdr #32: fails to compile on debian

N4CNR, 2024-07-30:
flags gtk+-3.0`    -c -o src/version.o src/version.c
cc -pthread -o pihpsdr src/MacOS.o src/about_menu.o src/actions.o src/action_dialog.o src/agc_menu.o src/ant_menu.o src/appearance.o src/band.o src/band_menu.o src/bandstack_menu.o src/configure.o src/css.o src/cw_menu.o src/discovered.o src/discovery.o src/display_menu.o src/diversity_menu.o src/encoder_menu.o src/equalizer_menu.o src/exit_menu.o src/ext.o src/fft_menu.o src/filter.o src/filter_menu.o src/gpio.o src/iambic.o src/i2c.o src/led.o src/main.o src/message.o src/meter.o src/meter_menu.o src/mode.o src/mode_menu.o src/mystring.o src/new_discovery.o src/new_menu.o src/new_protocol.o src/noise_menu.o src/oc_menu.o src/old_discovery.o src/old_protocol.o src/pa_menu.o src/property.o src/protocols.o src/ps_menu.o src/radio.o src/radio_menu.o src/receiver.o src/rigctl.o src/rigctl_menu.o src/rx_menu.o src/rx_panadapter.o src/screen_menu.o src/sintab.o src/sliders.o src/startup.o src/store.o src/store_menu.o src/switch_menu.o src/toolbar.o src/toolbar_menu.o src/transmitter.o src/tx_menu.o src/tx_panadapter.o src/version.o src/vfo.o src/vfo_menu.o src/vox.o src/vox_menu.o src/xvtr_menu.o src/waterfall.o src/zoompan.o src/pulseaudio.o  src/soapy_discovery.o src/soapy_protocol.o \
	src/alsa_midi.o src/midi2.o src/midi3.o src/midi_menu.o   src/saturndrivers.o src/saturnregisters.o src/saturnserver.o src/saturnmain.o src/saturn_menu.o  -lpulse-simple -lpulse -lpulse-mainloop-glib  `pkg-config --libs gtk+-3.0`  -lSoapySDR  -lasound -lwdsp -lm -lrt
/usr/bin/ld: src/noise_menu.o: in function `set_noise':
noise_menu.c:(.text+0x29c): undefined reference to `SetRXARNNRRun'
/usr/bin/ld: noise_menu.c:(.text+0x2ac): undefined reference to `SetRXASBNRreductionAmount'
/usr/bin/ld: noise_menu.c:(.text+0x2bc): undefined reference to `SetRXASBNRsmoothingFactor'
/usr/bin/ld: noise_menu.c:(.text+0x2cc): undefined reference to `SetRXASBNRwhiteningFactor'
/usr/bin/ld: noise_menu.c:(.text+0x2dc): undefined reference to `SetRXASBNRnoiseRescale'
/usr/bin/ld: noise_menu.c:(.text+0x2ec): undefined reference to `SetRXASBNRpostFilterThreshold'
/usr/bin/ld: noise_menu.c:(.text+0x304): undefined reference to `SetRXASBNRRun'
/usr/bin/ld: src/receiver.o: in function `create_receiver':
receiver.c:(.text+0x3078): undefined reference to `SetRXARNNRRun'
/usr/bin/ld: receiver.c:(.text+0x3084): undefined reference to `SetRXASBNRreductionAmount'
/usr/bin/ld: receiver.c:(.text+0x3090): undefined reference to `SetRXASBNRsmoothingFactor'
/usr/bin/ld: receiver.c:(.text+0x309c): undefined reference to `SetRXASBNRwhiteningFactor'
/usr/bin/ld: receiver.c:(.text+0x30a8): undefined reference to `SetRXASBNRnoiseRescale'
/usr/bin/ld: receiver.c:(.text+0x30b4): undefined reference to `SetRXASBNRpostFilterThreshold'
/usr/bin/ld: receiver.c:(.text+0x30c8): undefined reference to `SetRXASBNRRun'
collect2: error: ld returned 1 exit status
make: *** [GNUMakefile:662: pihpsdr] Error 1

N4CNR, 2024-07-30:
EXTENDED_NR=ON is the issue. I will have to look other wdsp.

dl1ycf, 2024-08-21:
I suggest NOT to use extended nr. There is a modified WDSP by somebody which you can find somewhere. Better
try the current source since it has been updated to WDSP 1.25 and feature the new "trained" NR2.

closed 2024-08-21 by dl1ycf