dl1ycf/pihpsdr #38: Segmentation fault on Ubuntu 24.04

frspin, 2024-09-27:
Latest version of pihpsdr from github compile and run on my Ubuntu 24.04.
Hermes Lite is found but, after pushing "Start" I get a segmentation fault.

Running it with gdb give:

Thread 1 "pihpsdr" received signal SIGSEGV, Segmentation fault.
0x00007ffff7953e3d in gtk_widget_destroy ()
   from /lib/x86_64-linux-gnu/libgtk-3.so.0
(gdb) bt
#0  0x00007ffff7953e3d in gtk_widget_destroy ()
    at /lib/x86_64-linux-gnu/libgtk-3.so.0
#1  0x00005555555aec81 in tx_create_transmitter ()
#2  0x000055555558b733 in radio_start_radio ()
#3  0x000055555556dd5d in ext_start_radio ()
#4  0x00007ffff719e522 in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff719d48e in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#6  0x00007ffff71fc717 in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#7  0x00007ffff719ca53 in g_main_context_iteration ()
    at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#8  0x00007ffff73d388d in g_application_run ()
    at /lib/x86_64-linux-gnu/libgio-2.0.so.0
#9  0x0000555555560a85 in main ()

Regards

Franco Spinelli
IW2DHW

dl1ycf, 2024-10-04:
Thanks for the hint. It was a missing initialization which now matters after I closed a memory leak. It should be corrected now. Unfortunately the error did not show up on at least two other test systems (possibly they implicitly clear newly
allocated data structures).

closed 2024-10-04 by dl1ycf

frspin, 2024-10-04:
Il 04/10/24 09:45, dl1ycf ha scritto:
> Thanks for the hint. It was a missing initialization which now matters after I closed a memory leak. 
> It should be corrected now. Unfortunately the error did not show up on at least two other test 
> systems (possibly they implicitly clear newly
> allocated data structures).

I can confirm that problem is solved.

Thanks

Franco Spinelli
IW2DHW