dl1ycf/pihpsdr #39: Mute not working

mcogoni, 2024-10-11:
Hi Christoph,
I wasn't able to use the mute function: it does nothing at all.

marco / IS0KYB

dl1ycf, 2024-10-28:
The mute function is intended to mute the audio that goes to the radio. Headphones connected to sound cards that
are connected directly to the computer will not be affected.

So, be more specific in your problem report, perhaps piHPDSDR behaves according to the manual, but not according to your expectations?

dl1ycf, 2024-10-28:
I have made this more clear in the manual now. But the behaviour is as I said: only auto to the radio is affected. If the radio has no codec then these mute commands have no effect.

dl1ycf, 2024-10-28:
I totally disagree on your view of the MUTE function. If it is equivalent to moving the AF level slider to the left no one needs it. The intended use is to mute built-in radio speakers while doing digi mode.

closed 2024-10-28 by dl1ycf

mcogoni, 2024-10-28:
The mute button exists in every audio appliance, from tvs to radios and hifi amplifiers, in order to quickly and temporarily silence the sound. You don't want to use the AF volume because you don't want to lose your set volume and it'd be slower anyway.

dl1ycf, 2024-10-28:
You can set the AF slider to zero, it will only affect the active receiver. The current behaviour is exactly as I want it to have, whether you like it or not.