dl1ycf/pihpsdr #41: TX drive

PE1PWR, 2024-10-28:
Hi Christoph.
Keep up the good work.
I have one question.
Why wont work the TC drive correctly.
I cant get the signal from zero to max. I have turn on TX drive and set to 1.
Ask also Johan PA3GSB about it and he say to me that is works perfect. I have try yours PIHPSDR version and the version from Johan site. I have 3 Radioberry's and results is the same. No TX drive from zero to max. 
For a TX drive power I use now 4 pin diodes and potmeter for level TX power. Question why works software TX drive within Pihpsdr so bad.
Wim
PE1PWR
73s

dl1ycf, 2024-10-28:
It all works OK, but I have difficulties to understand your email. If you have no TX power out, have you adjusted the calibration values in the PA menu? The TX level actually sent to the radio depends both on the TX drive slider and the
PA calibration value. If the latter is too high, you will have zero RF output.
So set the drive slider to 100, then check "TUNE use drive" in the TX menu, then go TUNE, and then slowly
decrease the PA calibration value of the band you are working in  until you get nominal output power.

BTW it is all in the manual.

PE1PWR, 2024-10-28:
Sorry for my bad Englisch Christoph.
Thanks for quick answering. 
TX out works. I have a full build Radioberry. That's not the problem. Only problem is drive slider works not right to set your output power for my amp. I can get the tx power to minimum. Say 1 watt for FT 8. I have build also a transmitter based on a Red Pitaya SDR. TX drive slider works fine and I can also tune PA so that every channel has the same maximum power. 
I have test Radioberry with other software and I can set the right output power I want to use. I hope you understand my question. Why won't works the drive slider.

dl1ycf, 2024-10-29:
No I do not understand. Produce a point-by-point list what happens. "works not right" is not sufficient.
Note that the hardware of the radioberry as a step-attenuator in the TX that has very few steps. So the HPSDR simulator
software must translate the TX drive value (something between 0 and 255) to one of the 15 available steps. piHPSDR does this for a HL2 but if the radio berry does not identify itself as a HL2 it has to take care of this itself.

dl1ycf, 2024-10-29:
BTW minimum power should not be 1 watt but 0 watt. This probably comes from the limitations of the hardware which can only do TX attenuation between 0 and -7.5 dB. If a HL2 is detected, piHPSDR does additional scaling of the IQ samples
to achieve smooth regulation from zero to full power. But the whole story is to get along with the limitations of the hardware of the modem chip.

dl1ycf, 2024-10-29:
WIth HL2 I refer to the "HermesLilte-II" which has the same modem chip as the radioberry.
Somewhere I read the Radioberry uses the same FPGA code than the HL2 but of course it must then also be
detected as an HL2, this is written in the top bar of the pihpsdr window.

And perhaps you should state what exactly happens if you slowly move the drive slider from 0 to 100, you never did, but then you complain it does not work as expected. With such a ,,problem report'' there can be no help.

dl1ycf, 2024-11-21:
Issue closed since I have not heard anything in 3 weekss

closed 2024-11-21 by dl1ycf