dl1ycf/pihpsdr #42: Save Sound resource with Mode Profile

K7MDL2, 2024-11-06:
Request to add the current sound device (and the local checkbox) with each Mode profile.

Example:
If in USB mode, the audio output is likely to go to speakers.  TX from Mic.
If in DigU audio out should be the local loopbacks. 

This provides seamless transition between Dig and non-Dig modes.  The Mode Profiles already have the rest of the key parameters saved and restored per mode.

dl1ycf, 2024-11-21:
If you configure your output soundcard in CW, then it would be quite in-convenient if you loose this setting when going to SSB. What is already there:
- you can use a mic connected to a radio  while there is a "local microphone" used for digi.
- you can use headphone connected to radio while there is "local rx audio" used for digi

And on many systems, the virtual cables can  be part of a "patch panel" so it is no problem to have Mic/Headphone
and DIGI operating at the same time.

closed 2024-11-21 by dl1ycf

K7MDL2, 2024-11-21:
The DigiU/DigiL modes are in a different settings group from SSB.   If SSB and CW are in different groups then they can simply be set to the same sources as they will be returned each mode change.

The issue I get with trying to use the output monitor is the digiU mode requires the AFGain to be adjusted and that forces the speaker/headphone to be the wrong level for listening.  This forces external volume management were one constantly affects the other, on a 7" screen where piHPSDR is fullscreen, messing about with an extra UI fiddling with 2 levels for the same source is impractical.

Thanks for looking at this.

  *   Mike

dl1ycf, 2024-12-02:
Let me explain one principle: I NEVER program any "automatic" change in piHPSDR which might possibly be harmful or even destructive. Therefore, updating the TX-drive automatically upon a band change  (to that value used before on the band) will not be included since this may blow your PA if you forgot that you worked on that band full power (without PA) previously.  Likewise, automatic AF volume changes will not be included since this may damage your ears. The principle is simply as follows: if you damage your PA or your ears, this is because YOU have moved a slider and not piHPSDR.

So, get a headphone with a volume regulator, adjust AF output for the digimode program and then adjust the headphone for pleasant listening.

Having said this, including the "Mic Gain" in the mode-depending settings is OK, and I think I will do this. Typically, the MicGain slider is at 0 dB for digi modes but it may be required to have it at some positive value if the mic preamp is too weak.