dl1ycf/pihpsdr #43: Support for windows

Roturbo, 2024-11-08:
Hello,

i have test this version with windows support, it uses 1/3 or less CPU than Thetis but is outdated 
and for example on Anan 100D or 200D is not possible to get full power.
https://github.com/ew8bak/pihpsdr 

It is a nice option for machines with less power, small windows tables with touch etc, but now i see this new updated version
with many improvements on this video https://www.youtube.com/watch?v=SpwmlVEWeYI&ab_channel=RichTerrusa 
and i wander if it is possible to also add windows support for this version?

Than the user just need to change the file with screen size before compile it, this way getting a full screen pihpsdr 
running on is windows machine.

It is a suggestion but i don´t know how difficult it is to implement the windows support.

Regards

ydalton, 2024-11-20:
If you want to run the application on Windows, your best bet would be to use WSL.

dl1ycf, 2024-11-21:
If you have a running Windows "MinGW" environment with GTK support, just replace the pihpsdr sources you have
by the latest version and it should work.  WDSP is now  integrated into the piHSDPR source code tree,so no
need to compile a wdsp.dll separately.
This is all I can say since my lab is a windows-free zone.

closed 2024-11-21 by dl1ycf

Roturbo, 2024-11-21:
Tanks for your replay @dl1ycf 

i will try to do that and see if it works.

Regards

ydalton, 2024-11-21:
> If you have a running Windows "MinGW" environment with GTK support, just replace the pihpsdr sources you have by the latest version and it should work. WDSP is now integrated into the piHSDPR source [...]

I'm sorry, but what about the Linux/Mac specific elements such as sound? I see the code has a PortAudio backend alongside PulseAudio, but will it work on Windows?

Roturbo, 2024-11-21:
@ydalton   please check the  link 

https://github.com/ew8bak/pihpsdr  

it explains the prerequisites includes installing the pulseaudio and portaudio and i confirm it works,
tested with Anan 100D and Hermes Lite 2.