dl1ycf/pihpsdr #47: Radioberry

Marc1950, 2024-12-17:
Hello
I am looking for the method to make PIHPSDR version DL1YCF work with a RadioBerry.
73

dl1bz, 2024-12-18:
There are a lot of solutions you can found in the internet, how to make the Radioberry ready and work. It should be detected as Hermes Lite V2 with HPSDR protcol 1, the Radioberry if running will bind to the ethernet interface of the Pi. As a first check I would try with SDR console from a WIN PC, if you can found and control the radioberry via ethernet. After this, go back to Linux:
Maybe you need to compile piHPSDR **without** GPIO support, because (if I remember right) there are conflicts with some GPIO pins they used by the Radioberry HAT for the Pi too. Otherwise you can change/adjust the GPIO assignment of piHPSDR in its source code.

73 Heiko, DL1BZ

dl1ycf, 2025-01-05:
Since I do not own a radioberry device, and got no information thereon in the last five years, there is not much I can offer here.

closed 2025-01-05 by dl1ycf