dl1ycf/pihpsdr #49: Pihpsdr > Pluto >RTLsdr > Hackrf

Marc1950, 2025-01-08:
hello
I installed Pihpsdr with a Pluto but unfortunately it does not find the hardware at startup and the same with an RTL key and a Hackrf.
The libraries are well detected with the command "SoapySDRUtil -find"
in which direction should I look ???
My Rapsberry 4 configuration with a 7 inch screen
73 by Marc F6GKG

dl1ycf, 2025-01-15:
For me it runs with pluto and an RF stick. Have you created a file "make.config.pihpsdr" in the pihpsdr directory and put the line "SOAPYSDR=ON" there? Whether the SOAPY option is included is indicated in the first screen where the compile time options reported there must contain SOAPY. This is all explained in detail in Appendix G of the manual.

Furthermore, the *libraries* are listed with "SoapySDRUtil -info" and the available *devices* with "SoapySDRUtil -find".

Marc1950, 2025-01-15:
hello dear om
thanks for the answer.
no I did not create this file. on the other hand the Soapy libraries are
present.
I am not at home at the moment, I will see when I return and I will inform
you.
73

Marc1950, 2025-01-21:
Hello
Forgive my ignorance but I did not understand for the file to create in pihpsdr, what extension to put, I tried to create it with the console but it does not work.
on the other hand I modified the Makefile by putting SoapySDR on and recompiled and it works. I am very disappointed with the HackRF, a lot of noise and significant parasitic lines on the low bands.
73 F6GKG

Richz28, 2025-01-22:
Hi Marc , what are you trying to do with pihpsdr, what operating system, fresh install ,or update, what type of radio do you want to configure?

What think it sounds like your trying to was update Pihpsdr, and you had a error, and update didn't complete,
So please try doing this in order.

cd ~/github/pihpsdr

Make clean

Git pull

LINUX/libinstall.sh<http://libinstall.sh/>

make

./Pihpsdr

You may have to put all your settings back , for each menu, because one of these updates' generated new scr files to default settings, no big deal, Christopher has done a great job with maintaining, updates and added new features to Pihpsdr,  no easy task, be grateful we have him, but be patient, don't update the Pihpsdr because a friend says to or just for the sake of keeping up to date, sometimes updates need to be proven and tryed by experienced Linux system operates, only update when something is not working, or until new releases are posted, just until your Linux skills are improved, don't feel bad took me a year and still learning, hope this helps if not I am willing to give it another go, ttyl Rich Terrusa kc2zur 73

dl1ycf, 2025-01-22:
Before you run the pihpsdr LINUX/libinstall.sh script, make sure that you remove ALL SoapySDR components from your system  that you may have loaded before. Since the SoapySDR version in the standard repositories is version0.7, but piHPSDR requires version0.8, left-over 0.7 components just lead to "no go". In standard "virgin" operating systems no SoapySDR components are loaded, but just in case you played with it before: uninstall them.
The piHPDSR LINUX/libinstall script then will install the SoapySDR core library but no radio-specific SoapySDR modules. Here, there are install scripts now in LINUX/libinstall.sh (this separation has been made last week) for several radios such as AdalmPluto, RTL sticks, AirSpy and SDRPlay. Support for other radios must be compiled from the sources and installed in /usr/local, for the reasons explained above you cannot use modules from the repos since they are in the "version0.7 universe".

Marc1950, 2025-01-22:
Ok thanks for the details.
What I'm interested in is using your version of Pihpsdr with a HackRF. 
Where can I find the sources for this radio?

dl1ycf, 2025-01-22:
Asking Dr. Google for "hackrf soapy" gave

https://github.com/pothosware/SoapyHackRF

If this is already too difficult for you, then I wish you the very best.

dl1ycf, 2025-01-24:
Using stuff I found on the page mentioned above, I tentatively made a script named soapy.hackrf.sh in the LINUX directory. It compiles well and installs the HackRF SoapySDR module, but of course I did not test it because I do now own a HackRF.

If you update your pihpdsr to the latest version, do the following

cd <your pihpsdr directory>
git pull
LINUX/soapy.hackrf.sh

and this should install the module (recompilation of pihpsdr is not needed but also would do  no harm).
Then the command

SoapySDRUtil -info

should list libHackRFSupport in the list of modules. Further, if a HackRF is attached to the computer,

SoapySDRUtil -find

should find it.

Marc1950, 2025-01-24:
good evening
thank you very much for the info.
i will try that tomorrow
73

Marc1950, 2025-01-25:
Hello
Sorry but the update did not add soapy.hackrf in Linux only pluto and RTL.

dl1ycf, 2025-01-25:
I do not like my leg to be pulled. You can see the "missing" file in the LINUX directory here on github.
If it is not on your machine your "git pull" failed.

dl1ycf, 2025-01-25:
The only explanation I have that you use the TEST branch. If there is no test currently underway, it is not
kept in sync with the master branch. But I will now synchronize it. If you are on the TEST branch, pse switch
back to master.

Marc1950, 2025-01-25:
I'm sorry but I don't understand anything you're telling me in your last message.
You're not to blame because I don't understand anything about programming.
I regret having started this project I thought it was more "plug and play". It cost me money and time. Another assembly that will end up in a closet.
In any case to go back the lines:
cd
git pull
LINUX/soapy.hackrf.sh
give me an error message in pihpsdr.
Sorry to waste your time and think we'll stop there.
73

dl1ycf, 2025-01-25:
The lines you quoted cannot work. The first statement must be "cd $HOME/pihpsdr", or alternatively, after "cd"
the second statement should be "cd pihpsdr".

So the "git pull" already should have flagged an error that you are not in a git repo.

But you are right, you should seek help from a local ham with some experience with such things.

closed 2025-01-25 by dl1ycf