dl1ycf/pihpsdr #50: HW AGC setting not saved by radio

guitarpicva, 2025-01-18:
Would this be possible?  The Airspy HF+ Dual has HW AGC and provides much more signal level when activated.

Is there any way to save that setting, or must it be set for each radio device when started?

Lastly would this be a better issue for the upstream repo?

guitarpicva, 2025-01-18:
Also noticed that it is not correctly displayed when going back to the dialog for "Radio".

So if I go and turn it on manually and it is working, close the dialog and then re-open the dialog, it shows it disabled, even though it is obviously turned on.

guitarpicva, 2025-01-24:
> Will do Christoph.
>
> Thanks for the quick response.
>
> Mitch
>
> Mitch, AB4MW
>
> Sent with [Proton Mail](https://proton.me/mail/home) secure email.

That control tracks properly now. Thank you!

dl1ycf, 2025-01-24:
So I can close this issue.

closed 2025-01-24 by dl1ycf