dl1ycf/pihpsdr #54: Parameter for bypass discovery

frspin, 2025-02-13:
Please add a parameter for bypass discovery step when an IP of device is known.

As example:

pihpsdr -d xxx.xxx.xxx.xxx

start directly device at xxx.xxx.xxx.xxx without discovery and any other question.

Regards

Franco Spinelli
IW2DHW

dl1ycf, 2025-03-01:
Exchanging a discovery packet is necessary to get same basic information from the radio, such as its MAC address and its board type. So I am not sure what you want, perhaps you look for the "Autostart" option in the protocols menu? Then you have discovery but no questions.

closed 2025-03-01 by dl1ycf

dl1ycf, 2025-03-04:
While a "routed discovery" is implemented in piHPSDR (insert IP addr in "Radio IP addr" text field), this is against the HPSDR standard which *requires* broadcast discovery packages. "Routed" discovery can be used across network boundaries, and works e.g. for the HermesLite-II.
However piHPSDR will still query existing networks by default, so what you are after is a "No broadcast discovery" option in the "protocols" menu? You can easily implement this, but I am sure if I do this, there will be a hell of
support questions, so I refrain therefrom.

frspin, 2025-03-04:
Il 04/03/25 16:11, dl1ycf ha scritto:
> While a "routed discovery" is implemented in piHPSDR (insert IP addr in "Radio IP addr" text field),
> this is against the HPSDR standard which /requires/ broadcast discovery packages. "Routed" [...]

Inserted IP address in Radio IP addr IP is not saved and, next time I start pihpsdr, broadcast 
discovery packages are send in both my interfaces.
I have 2 network interfaces: one for normal network work and one dedicated to HL2.

What I will is a broadcast to only IP defined in Radio IP addr field and a save for this parameter.
This, with "autostart if only one devices" can give a more fast startup of program.
This is how Quisk is working.

Obviously if one have more than a HL2 can not define a IP and the normal discovery will be done.

Franco Spinelli
IW2DHW

dl1ycf, 2025-03-04:
>
> Inserted IP address in Radio IP addr IP is not saved

Are you sure? Once the text  field contains a valid IP addr it  is saved.
But this is mainly meant for connecting  to a radio through a VPN.

Try to put at the top of old_discovery.c, function old_discovery()

  discover(NULL, 2);
  return;

This should  ONLY try a "directed UDP  discovery" to the ip addr specified.

dl1ycf, 2025-03-09:
I have now changed the discovery process in two respects:

- try fixed IP address first, loop through interfaces later
- when using a fixed IP address, abort discovery when one valid response has been received
- when a valid response has been received from a fixed IP address, do not loop through the interfaces.

This should now shorten down the discovery for you to well below 1 second.

Please try and report.

frspin, 2025-03-09:
Il 09/03/25 18:51, dl1ycf ha scritto:
> I have now changed the discovery process in two respects:
>
>   * try fixed IP address first, loop through interfaces later
>   * when using a fixed IP address, abort discovery when one valid response [...]

Updated to latest GIT.
Defined only protocol 1 discovery and autostart.

Protocols.props is:

property_version=3,00
autostart=1
enable_saturn_xdma=1
enable_usbozy=1
enable_stemlab=1
enable_soapy_protocol=0
enable_protocol_2=0
enable_protocol_1=1

When starting pihpsdr this is discovery done:

0,506 loadProperties: protocols.props
0,608 old_discovery
0,608 discover: looking for HPSDR device with IP 169.254.19.221
0,608 discover_receive_thread
0,608 old_discovery: received 60 bytes
0,608 discovered HL2: Gateware Major Version=73 Minor Version=2
0,608 old_discovery: name=HermesLite V2 min=0,000 MHz max=38,400 MHz
0,608 old_discovery: found device=506 software_version=732 status=2 address=169.254.19.221 
(00:1C:C0:A2:13:DD) on  min=0,000 MHz max=38,400 MHz
0,608 discovery: exiting discover_receive_thread
0,608 discover: exiting HPSDR discover for IP 169.254.19.221
0,608 Trying to detect via TCP with IP 169.254.19.221
3,611 discover: select() timed out on TCP discovery socket
3,611 discovery found 1 devices
3,611 discovery: found device=506 software_version=732 status=2 address=169.254.19.221 
(00:1C:C0:A2:13:DD) on UDP
3,714 discovery: found 1 devices
3,714 discovery: found protocol=0 device=506 software_version=732 status=2 address=169.254.19.221 
(00:1C:C0:A2:13:DD) on UDP
3,718 Device Protocol=0 name=HermesLite V2
3,719 loadProperties: remote.props
3,719 current host: 127.0.0.1:50000
3,727 HOST ENTRY #0 = 127.0.0.1:50000
3,730 loadProperties: gpio.props
3,730 gpio_set_defaults: 0
3,730 gpio_set_defaults: 0
3,755 showing device dialog
3,755 discovery: devices=1 autostart=0

Is this correct?

Now there are 5.4 second from launch and when window show up

Regards

Franco Spinelli
IW2DHW

dl1ycf, 2025-03-10:
I guess you have not enabled SATURN, USBOZY and STEMLAB as compile-time options so these protocols stay
"enabled" in protocol.props but this has no effect (and the log file shows only Protocol1 is tried).

Usually I try both UDP and TCP if there is a fixed IP address, and in your case the HL2 detection only took 1 msec:

0,608 discover: looking for HPSDR device with IP 169.254.19.221
0,608 discover_receive_thread
0,608 old_discovery: received 60 bytes
0,608 discovered HL2: Gateware Major Version=73 Minor Version=2
0,608 old_discovery: name=HermesLite V2 min=0,000 MHz max=38,400 MHz
0,608 old_discovery: found device=506 software_version=732 status=2 address=169.254.19.221
(00:1C:C0:A2:13:DD) on  min=0,000 MHz max=38,400 MHz
0,608 discovery: exiting discover_receive_thread

 but then 3 sec wait time for TCP. 

0,608 Trying to detect via TCP with IP 169.254.19.221
3,611 discover: select() timed out on TCP discovery socket

Normally a TCP connect should fail immediately if the port is not open, but the HL2 firmware is not aware of TCP and therefore the TCP connection runs into a time-out. To resolve this issue, I have also added a "Enable TCP" check-box which should be UN-checked for the HL2 (Note the ip addr now went  into the file "ipaddr.props" to be more consistent with the rest of the program).

For me, the start of the radio with autostart is now very fast:

     1,307 old_discovery
     1,307 discover: looking for HPSDR device with IP 131.246.58.187
     1,307 discover_receive_thread
     1,307 old_discovery: received 60 bytes
     1,307 discovered HL2: Gateware Major Version=74 Minor Version=2
     1,307 old_discovery: name=HermesLite V2 min=0,000 MHz max=38,400 MHz
     1,307 old_discovery: found device=506 software_version=742 status=2 address=131.246.58.187 (00:1C:C0:A2:13:DD) on  min=0,000 MHz max=38,400 MHz
     1,307 discovery: exiting discover_receive_thread
     1,307 discover: exiting HPSDR discover for IP 131.246.58.187
     1,307 discovery found 1 devices
     1,307 discovery: found device=506 software_version=742 status=2 address=131.246.58.187 (00:1C:C0:A2:13:DD) on UDP
     1,410 discovery: found 1 devices
     1,410 discovery: found protocol=0 device=506 software_version=742 status=2 address=131.246.58.187 (00:1C:C0:A2:13:DD) on UDP
     1,411 Device Protocol=0 name=HermesLite V2
     1,412 loadProperties: remote.props
     1,412 current host: 127.0.0.1:50000
     1,414 HOST ENTRY #0 = 127.0.0.1:50000
     1,414 HOST ENTRY #1 = 131.246.58.177:50000
     1,415 loadProperties: gpio.props
     1,415 gpio_set_defaults: 0
     1,416 gpio_set_defaults: 0
     1,436 showing device dialog
     1,436 discovery: devices=1 autostart=1
     1,667 radio_start_radio: setup RECEIVERS protocol=0

So I would ask you to pull+make+try.

frspin, 2025-03-10:
Il 10/03/25 09:56, dl1ycf ha scritto:
> I guess you have not enabled SATURN, USBOZY and STEMLAB as compile-time options so these protocols stay
> "enabled" in protocol.props but this has no effect (and the log file shows only Protocol1 is [...]

Done and is working!

Many thanks

Franco Spinelli
IW2DHW