dl1ycf/pihpsdr #55: CAT/TCI Autoreport (on CAT TCP) does not send VFO A mode changes

guitarpicva, 2025-02-15:
CAT/TCI Autoreport (on CAT TCP) does not send VFO A mode changes.

Tested after a configuration change and restart of piHPSDR

RPi5 16 GB RAM
RPiOS Latest w/all updates current

dl1ycf, 2025-03-01:
At the moment, if you just turn autoreporting on in the menu, CAT TCP report *frequency* changes only, since this is what you
normally need for PAs. If you request autoreporting via CAT (AI or ZZAI commands), you can set higher "autoreporting" levels. At the moment, higher levels mean that VFO-A mode is reported but this requires an "AIx" command coming in with x>1.

So the question is: were mode changes explicitly request by an AI or ZZAI command? If not, they are not reported.

guitarpicva, 2025-03-01:
Not sure if that's documented.  I don't remember seeing it.  I will test with the CAT AIx command and see what happens.  Thank you for the explanation.

dl1ycf, 2025-03-04:
> Not sure if that's documented. I don't remember seeing it. I will test with the CAT AIx command and see what happens. Thank you for the explanation.

It is documented in the piHPSDR manual, do where do you have looked it up?

closed 2025-03-04 by dl1ycf