dl1ycf/pihpsdr #56: Add support caribou lite

heathhey, 2025-02-23:
I backed the caribou lite SDR hat for raspberry pi: https://www.crowdsupply.com/cariboulabs/cariboulite-rpi-hat

The software uses Soapy SDR. I built pihpsdr with soapy support. When I start pihpsdr, it does detect one of the radios (the S1G) device. When I start that, I don't get any info on the waterfall. It seems like there are just a few missing things that would enable piHPSDR to work with this device. I have some programming experience, and am looking for any guidance in adding this radio. I suspect it will be up to me, as the developers of pihpsdr don't have this hardware. Here is output from the discovery, I hope there are some clues that can be pointed out and where I might need to add things in the code to make this work. Thanks for all you work. I use piHPSDR with my HL2, and it works great. 

```
     0.000 Base priority on startup: 0
     0.000 Base priority after adjustment: 0
     0.000 startup: working directory not changed.
     0.117 Build: v2.5 (Commit: 6104fac, Date: 2025-02-19)
     0.117 GTK+ version 3.24.38
     0.117 sysname: Linux
     0.117 nodename: raspberrypi
     0.117 release: 6.6.51+rpt-rpi-v8
     0.117 version: #1 SMP PREEMPT Debian 1:6.6.51-1+rpt2 (2024-10-01)
     0.117 machine: aarch64
     0.118 load_css: failed to load CSS data from file default.css
     0.118 load_css: hard-wired CSS data successfully loaded
     0.118 load_font: CSS font set to FreeSans
     0.118 create top level window
     0.122 Monitor Number within Screen=0
     0.122 Monitor: width=1280 height=720
     0.122 display_width=800 display_height=480
     0.122 create image and icon
     0.122 Failed to open file âhpsdr.pngâ: No such file or directory
     0.122 Failed to open file â/usr/share/pihpsdr/hpsdr.pngâ: No such file or directory
     0.122 Failed to open file â/usr/local/share/pihpsdr/hpsdr.pngâ: No such file or directory
     0.123 create grid
     0.123 add grid
     0.123 add image to grid
     0.123 create pi label
     0.123 add pi label to grid
     0.123 create build label
     0.123 add build label to grid
     0.123 create status
     0.123 add status to grid
     0.236 g_idle_add: init
     0.248 init
     0.250 Securing wisdom file in directory: /home/heath/projects/pihpsdr/
     0.350 state_cb: 2
     0.350 audio: state_cb: PA_CONTEXT_AUTHORIZING
     0.365 state_cb: 3
     0.365 audio: state_cb: PA_CONTEXT_SETTING_NAME
     0.466 state_cb: 4
     0.466 audio: state_cb: PA_CONTEXT_READY
     0.474 Output: 69: alsa_output.platform-bcm2835_audio.stereo-fallback (Built-in Audio Stereo)
     0.575 Input: 69: alsa_output.platform-bcm2835_audio.stereo-fallback.monitor (Monitor of Built-in Audio Stereo)
     0.681 loadProperties: protocols.props
     0.885 old_discovery
     0.886 discover: looking for HPSDR devices on lo
     0.886 discover: bound to lo
     0.886 discover_receive_thread
     2.901 discovery: bytes read -1
     2.901 old_discovery: recvfrom socket failed for discover_receive_thread: Resource temporarily unavailable
     2.901 discovery: exiting discover_receive_thread
     2.901 discover: exiting discover for lo
     2.901 discover: looking for HPSDR devices on eth0
     2.901 discover: bound to eth0
     2.901 discover_receive_thread
     2.901 old_discovery: received 60 bytes
     2.901 discovered HL2: Gateware Major Version=72 Minor Version=8
     2.901 old_discovery: name=HermesLite V2 min=0.000 MHz max=38.400 MHz
     2.902 old_discovery: found device=506 software_version=728 status=2 address=192.168.1.79 (00:1C:C0:A2:00:04) on eth0 min=0.000 MHz max=38.400 MHz
     4.917 discovery: bytes read -1
     4.917 old_discovery: recvfrom socket failed for discover_receive_thread: Resource temporarily unavailable
     4.917 discovery: exiting discover_receive_thread
     4.917 discover: exiting discover for eth0
     4.917 discover: looking for HPSDR devices on wlan0
     4.917 discover: bound to wlan0
     4.917 discover_receive_thread
     4.948 old_discovery: received 60 bytes
[... 171 lines cut ...]
    11.537 RX freq range [389.500000 MHz -> 510.000000 MHz step=0.000000]
    11.537 RX freq range [779.000000 MHz -> 1020.000000 MHz step=0.000000]
    11.537 RX antenna: TX/RX Sub1GHz
    11.537 TX antenna: TX/RX Sub1GHz
    11.537 has_automaic_gain=1
    11.537 has_automaic_dc_offset_correction=0
    11.537 RX format available: CS16
    11.537 RX format available: CS8
    11.537 RX format available: CF32
    11.537 RX format available: CF64
    11.537 RX gain Modem AGC
    11.537 RX gain available: Modem AGC, 0.000000 -> 69.000000 step=0.000000
    11.537 Modem PA     11.537 TX gain Modem PA, 0.000000 -> 31.000000 step=0.000000
    11.537 soapy_discovery: name=Cariboulite min=389.500 MHz max=510.000 MHz
    11.639 discovery: found 4 devices
    11.648 Device Protocol=0 name=HermesLite V2
    11.649 Device Protocol=0 name=HermesLite V2
    11.650 Device Protocol=2 name=Cariboulite
    11.651 Device Protocol=2 name=Cariboulite
    11.652 loadProperties: gpio.props
    11.652 gpio_set_defaults: 0
    11.652 gpio_set_defaults: 0
    11.704 showing device dialog
    11.704 discovery: devices=4 autostart=0
```

dl1ycf, 2025-03-01:
I just canot tell. piHPDSR has selected a sample rate of 768000  which it uses as the default for an "unkown" SoapySDR
device (the only exceptions are rtslsdr = 1535000 and radioberry = 48000).

Note that pihpsdr can only sample rates that are a multiple of 48000 and I do not want to run a resampler at that high sample rate.

I guess one could so something, but without having such a hardware on the desk, it is virtually impossible to manage.

closed 2025-03-04 by dl1ycf

heathhey, 2025-03-04:
Thanks for your feedback. Your comment did give me some hints and things to look at.