dl1ycf/pihpsdr #58: pihpsdr with rspdx r2

Monkeyman1066, 2025-03-19:
When I use my rspdx, everything seems to work but I get the following in the antenna section:

[image]

dl1ycf, 2025-03-24:
Look in the log file which antennas Soapy reports. For my Pluto it reads

     1,593 RX antenna: A_BALANCED
     1,604 TX antenna: A

and this is what is then seen in the menu. piHPSDR will (and should) use the names reported by SOAPY.
If the log file states correct names and the menu reports ugly ones, it is my fault. But first check the log file.

Monkeyman1066, 2025-03-24:
Thanks for the reply. I'll have to find out how to create/access a log file, I should manage.

I've checked soapy with : SoapySDRUtil --probe="driver=sdrplay" . Antennas are listed correctly.

I'll check the log file.

dl1ycf, 2025-03-24:
please quote the SOAPY discovery part of the log here or send it to dl1ycf@darc.de.  What is in the log file
following "RX antenna:" or "TX antenna:" should appear in the menu.

closed 2025-03-25 by dl1ycf