dl1ycf/pihpsdr #59: Macintosh dependencies

mpvano, 2025-03-19:
Hi! Keep up the great work!

Just a note that when following your macintosh os build instructions, I had some missing dependencies and had to use brew to add the following packages (these seemed to be the correct names):

portaudio
gtk+3

You might want to add/correct these - otherwise everything wen't perfectly on my Apple M2 Mini under Sequoia 15.3.2 with my Hermes Lite 2.

thanks again....

dl1ycf, 2025-03-24:
The MacOS/libinstall.sh script contains

################################################################
#
# All homebrew packages needed for pihpsdr (makedepend and
# cppcheck are useful for maintainers)
#
################################################################
$BREW install gtk+3
$BREW install librsvg
$BREW install pkg-config
$BREW install portaudio
$BREW install fftw
$BREW install libusb
$BREW install makedepend
$BREW install cppcheck
$BREW install openssl@3

so why were these missing for you?

dl1ycf, 2025-03-25:
If you switch from Intel Mac to Silicon Mac and use the "migration assistant", then it is best to delete homebrew completely and re-install it from scratch, since homebrew is in /usr/local for intel and in /opt/homebrew for M2. If homebrew was the only stuff installed in /usr/local, delete everything in /usr/local.