dl1ycf/pihpsdr #63: Feature request: adjust TX Filter Low/High settings for each mode group

g4dya, 2025-04-15:
It would help if the TX Filter High and TX Filter Low settings could be different for each mode group.

It is sometimes useful to filter signals from experimental modes (e.g. FreeDV RADE V1) while using DIGL/DIGU, but keep the default for LSB/USB.

This can be done by using the RX bandwidth, but this is not ideal operationally.

dl1ycf, 2025-06-10:
I think you refer to the RX/TX profiles. I have made a first shot at this, please try. I know it is not yet fully working
with the Client/Server model (this applies to the RX/TX profile as a whole, since changes are made on the client but
need to be stored in the profile on the server, I shall correct this  later).

g4dya, 2025-06-10:
I've tested this and it appears to work OK.

Thank you very much.

closed 2025-06-21 by dl1ycf