dl1ycf/pihpsdr #69: Where are settings files?

Hagtronics, 2025-07-25:
First of all: Wonderful work, well documented. Thank you for your hard work!  :-)

Version - 2.5 built from sources on July 22, 2025, running on Raspberry Pi 64 Bit, latest OS

I am trying to find the 'customization / configuration' files for backup. Specifically the MIDI, Function and Memory customization.

In the Manual on the 'Exit' menu (page 35) it says there should be a file called:  "XX-XX-XX-XX-XX-XX.props".

But for the life of me I cant find this - I can find three other ".props" files that list things like the radio and autostart behavior, but these do not contain the customization information. Searching the Raspberry PI thoroughly does not seem to turn up any other ".props" files.

Can you shed light on their location? Thank you.....

Richz28, 2025-07-25:
It's in PiHpSdr main folder, xxxxxxxxx stands for whatever your Mac address is, not xxxxxxxxx, hope this helps, Rich KC2ZUR 73

Hagtronics, 2025-07-26:
Thank you Richz28 - Yes it was right in front of me, 'hiding' in plain sight. Since I am using a SDRPlay device, what I am looking for is in the file: "sdrplay.props".

Case closed. I will go hide back under my rock now.  Have a good weekend.

closed 2025-07-26 by Hagtronics

Richz28, 2025-07-26:
Your welcome, it's all about team work, I had a learning period myself, never used piHpSdr with sdrplay , but according to tech minds, it makes for a great receiver, now since I never did this with sdrplay, someday I will,
When compile,
cd /github/Pihpsdr
make clean
git pull
./LINUX/libinstall.sh
./LINUX/soapy.sdrplay.sh

Then in PiHpSdr main folder open (make) file with a text editor, right click on it and make sure the line that says I believe (soapy) in it says =ON not =OFF,.     Use caps only for ON or OFF

The save file
Go back to terminal

cd github/pihpsdr
make

That's all should work, however I haven't tried this, so please try not to hold it against me,

Good luck , Rich KC2ZUR 73

Closed #69<https://github.com/dl1ycf/pihpsdr/issues/69> as completed.

Richz28, 2025-07-28:
> Hi Rich - I had no issue building the dl1ycf/pihpsdr - in fact out of all the receiver code out there, your instructions are about the ONLY ones that work first shot!
>
> I commented on the TechMinds video that he should look at this version as being far superior in user friendliness. :-)

Thank you